scooterhacking/scooterhacking-issues-public #39: Allow the motor to start, even if it's rolling backwards.

Nyami11, 2024-02-18:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Other

### Priority

low (take your time)

### SHUT Version

2.3.2

### SHFW Version

0.3.6

### Scooter model

Ninebot G30

### Description of the new feature

Right now, we are able to change the motor start speed from 0-10 kmph. It would be cool if we would set the start speed to 0, it would start the motor even if it's rolling backwards. This would be usefull for those who like to start the scooter from a complete stop like me, and sometimes when we do this we are accidentally rolling it backwards a bit, so it doesn't start. 🙂

### Contact Information

nyami11

PureComedi, 2024-02-18:
would be useful if there was a current limit implemented when the motor spun backwards to avoid FWD scooters from flipping on 30A brake curves and such

paulederbaus, 2024-02-18:
Scooter is now able to start from -3 km/h (0.3.7 prerelease V5_2 and higher)

closed 2024-02-18 by paulederbaus