scooterhacking/scooterhacking-issues-public #42: Feature: Firmware password locking

Nyami11, 2024-02-18:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Communication

### Priority

low (take your time)

### SHUT Version

0.3.7

### SHFW Version

2.3.2

### Scooter model

Ninebot G30

### Description of the new feature

Right now if a scooter is not connected to a phone, anyone can connect to it without any sort of pairing (on xiaomis at least you have to press the power button to pair), which poses a high risk of a bad actor connecting to it and modifing the settings in a way that it might even cause an accident. It would be great if we could put a password protection to scooters that would lock down firmware flashing, SHFW configuration managing, and scooter locking.

So how would it work? 
I think the best would be if we could set up a password that would be stored on the scooter, and SHUT would store it on the phone to make it more convinient. When we connect to it the scooter, SHUT would check if the scooter is password protected, and there would be some sort of autentication between SHUT and the scooter. Without authenticating, the scooter wouldn't allow the previously mentioned actions. So if it has password protecion, SHUT would ask for the password if it's not stored, and if it is stored, it would autenticate automatically. Third party dashboard apps like ninedash, m365tools should be able to connect the same way as they do now, but since they can't authenticate, the functions that require autentication wouldn't work in them. Storing a password instead of some sort of generated key would be the best option, since those can be remembered relative easily in case someone loses their phone for example.

Problems with it:

- Lock function still resets when someone unplugs the battery, so if someone steals it, they can just unplug-replug it to make it work again.
- A bad actor could still connect to a locked scooter with some un-authenticated application and stay in the bluetooth range, so you wouldn't be able to connect to it and unlock it.
- Biggest downside is obviously that if you forget your password, the only way to reset it is to reflash the firmware using an ST-Link V2.
- If you use a dashboard app, it would be a bit inconvenient, because you would have to disconenct with the dashboard app, then you would have to connect with SHUT just to lock it.
- I'm pretty sure implementing this feature would be pretty hard/time consuming, and there are much more important things to work on right now (like the G2 SHFW support).

### Contact Information

Discord: nyami11

paulederbaus, 2024-02-18:
@nopbxlr What do you think?

nopbxlr, 2024-02-18:
I'm not sure. Not only does it require a certain amount of time to put in place, but we would essentially give the users the ability to lock themselves out of their scooter (because inevitably people WILL forget their password/passcode), or by installing SHFW on someone else's scooter, to lock another user from their scooter. This kind of reminds me of the binding restrictions in the Segway app. I think CamiBLE's approach of BLE disabling after 15 seconds of no phone connected makes more sense, but we can't implement that. We have powerusers but we also have more limited ones. I wouldn't take that risk. The safety risk mentionned is not inherent to our modifiction anyways, it's due to the nature of the scooter.

nopbxlr, 2024-02-18:
There would be the possibility that we thought of in the past which is of adding a short passcode to simply change SHFW settings or even see that SHFW is installed at all (in a silent, no ACK manner if the passcode is invalid), but that would not prevent you from going back to another firmware and making that restriction disappear. It would just make it seem like stock firmware. Going as far as restricting a scooter with the above-mentioned issues is unfortunately something that we unanimously disagree on. Since this passcode idea has a different scope than a total lockout and doesn't aim at securing the device, but rather, like most physical locks "keep honest people out" (I hope you see the meaning of this), I will close the issue but keep that one in mind. Feel free to open one up if it ends up being forgotten, or if you wanna improve on the newly defined scope. We also hope to bring something like BLE timeout, or MAC address whitelist someday. Thanks!

closed 2024-02-18 by nopbxlr

oadq, 2024-09-05:
@nopbxlr 
I think this should be reconsidered.

The issues bxlr and Paul have pointed out can either be solved easily, or are not valid in my opinion.

1. Forgetting the password - same thing with forgetting your phones password, this is the users fault. There are many ways the user can screw something up currently with SHFW, disabling their e-brake, using stupidly high current values. It's impossible to make SHFW idiot proof. Show the user a warning. We shouldn't hold back on useful features just because there's a risk someone will mess it up, this only harms users.
2. Locking the owner out by installing SHFW and setting a password - the same thing can be accomplished already by installing the wrong firmware and bricking the scooter, in both cases, the user will need to flash with ST-Link. I'm not too familiar with BMS firmwares but technically you should be able to install a modified BMS firmware too, which can do some damage as well.

In my opinion the password lock should function in this way:
- we shouldn't try to prevent a "bad person" from unlocking the scooter, this is quite useless, use a physical lock to prevent theft
- instead, lock the SHFW config and firmware updates behind a password
- not locking everything out also solves the issue with dashboard apps, especially on iPhones
- this would not try to prevent stealing, but prevent a malicious actor from changing the SHFW config in a dangerous way silently, which can lead to the scooter owner getting in to an accident, which a physical lock cannot prevent

SchattenWolf2008, 2024-11-26:
Hey I also like the idea of adding a password feature.

I like the idea of having a mix of both, without the password it acts as normal firmware, and with the password it allows you to access SHFW and change the settings.

I understand the concern of people being able to lock others out, but I could also just connect to someone's scooter and flash bad firmware on it. A password could give me the option to prevent that.

On the other side, a password should not be a mandatory feature, I think its best to give the option to activate or deactivate it if you want, and a warning that informs you about the risk.

I really think a password would be a good feature, and I think the problems mentioned are not really big enough to not give the option at all. Especially as it could also just be a simple four digit password that you can easily remember.

It would be a nice feature to have, thank you for reading this.

Z1mmb0, 2024-11-27:
This topic never gets old 😁 the devs will hang us for commenting on this refused ticket.

I was not that big of a friend for the password solution, though i understand the concerns. Actually we now have customizable profile switch combinations. Also I remember a discussion about having to press, e.g. the brake, for 10 seconds in order to make the esc flashable.

Based on this there might be new solution paths, like a customizable combination for deactivating and activating the bluetooth module which state is persisted in a last mode. 
Over time the fruits start to hang lower and lower 😁😁

dobrishinov, 2025-02-21:
I also noticed that the Bluetooth connection is completely open and unprotected, which is a serious security issue. Anyone can connect and change settings, update the firmware, lock, or even brick the scooter.

A great solution would be to automatically disable BLE after a few seconds of inactivity or allow the user to manually toggle it on and off using a specific brake and throttle combination. This would give users better control over Bluetooth access.

I usually ride without my phone connected, and I often forget to turn off the scooter while charging. In these situations, anyone nearby could connect and tamper with the settings. Adding an option to disable BLE or restrict connections would greatly improve security.

I really hope this feature can be implemented. Thanks! ✌️