scooterhacking/scooterhacking-issues-public #43: Battery bars display bug

nopbxlr, 2024-02-18:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

UI / UX fix

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

SHUT v2.3.2 and all RC

### SHFW Version

SHFW v0.3.6 and all 0.3.7 pre releases

### Scooter model

Ninebot G30

### Description of the Bug

When using an alternative battery bar data value, the display goes back to battery level on idle. Furthermore, when % of current limit is used, braking is shown as 100%.

### Steps to Reproduce

Use a battery bar data value different than the default one, preferably % of max current. 

### Expected vs. Actual Behavior

This is already explained above

### Contact Information

@bxlr

paulederbaus, 2024-02-21:
Fix planend with reversing of Mr.tick

paulederbaus, 2024-02-22:
fix added - will be available from prerelease 0.3.7 v7

closed 2024-02-22 by paulederbaus