scooterhacking/scooterhacking-issues-public #44: Cruise control kick in only after 50% throttle. M365 regular latest shfw

AlexS-hub, 2024-02-20:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.3.2

### SHFW Version

0.3.6

### Scooter model

Xiaomi M365

### Description of the Bug

All modes of cruise control activate only with 50% throttle or more.

### Steps to Reproduce

Turn on scooter apply 30 throttle, no cruise control

### Expected vs. Actual Behavior

Cruise control kicks in, it doesn't

### Contact Information

artemlitovskikh27@yandex.ru

Nyami11, 2024-02-21:
> Scooter model
Xiaomi M365

Bit more context? Xiaomi M365 isn't even supported natively my SHFW. Do you have a 4-dot dash with a modified BLE firmware, a pro 1, or some other model's dash?

> Steps to Reproduce
Turn on scooter apply 30 throttle, no cruise control

What does "apply 30 throttle" even means?

> Expected vs. Actual Behavior
Cruise control kicks in, it doesn't

You mean, it activates, but doesn't give power?
There is a well known bug with cruise control, that sometimes (especially if you have a phone connected to it) it activates but doesn't give any power, so you have to turn it off, and re-enable it.

paulederbaus, 2024-02-21:
How are you using cruise control?

If you use time, the throttle value needs to be above the min value of hall sensor settings.

If you are using single / double tap, the value is hardcoded to 100. This means it depends on your throttle. With a good working throttle, this should be around 40%.
(A good working throttle goes from 40 to 190.)

AlexS-hub, 2024-02-21:
I use time mode. With cfw it works like a charm at about 15% -100%.  But with shfw cc only start from 50% -100%. I am sticking with cfw for now until it get fixed.

-- Отправлено из мобильного приложения Яндекс Почты

AlexS-hub, 2024-02-21:
Power based cc also has 50%-100% throttle activation range

paulederbaus, 2024-02-21:
If you dm me your uid via discord (paulederbaus), I can assign you an improved version.

AlexS-hub, 2024-02-21:
> If you dm me your uid via discord (paulederbaus), I can assign you an improved version.

You mean that you have ble that works correctly?

lekrsu, 2024-02-21:
> > If you dm me your uid via discord (paulederbaus), I can assign you an improved version.
> 
> You mean that you have ble that works correctly?

Why are we mixing BLE into this now?

paulederbaus, 2024-02-22:
Issue closed - small adjustment of throttle diff and speed diff per loop

closed 2024-02-22 by paulederbaus

AlexS-hub, 2024-02-22:
> Issue closed - small adjustment of throttle diff and speed diff per loop

It means i can flash shfw now and it will work like it should?