scooterhacking/scooterhacking-issues-public #45: BMS limit emulation

PureComedi, 2024-02-22:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Performance

### Priority

medium (upcoming releases)

### SHUT Version

2.3.2

### SHFW Version

0.3.7 

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

Implementation of a virtual limit that doesn't allow total current draw (including field weakening and acceleration boost) to go above a set limit.
Could be added as a 3rd current limit slider or replace Ld and Lq ideally. 

### Contact Information

purecomedi

paulederbaus, 2024-02-22:
Technically possible, but this is already available as two sliders and more customisable this way
-> maybe a total current display in shut would be helpful that summarises both sliders

(Changed to low)

PureComedi, 2024-02-22:
Issue is that flux to bms amp conversion can't be just straight up calculated without tests

paulederbaus, 2024-02-22:
> Issue is that flux to bms amp conversion can't be just straight up calculated without tests

This is not true. There is a very precise formula: I_total = sqrt(Iq² + Id²). This should match the current consumption very close.

PureComedi, 2024-03-04:
Today i found out that that formula and the 2 sliders indeed can't be properly used to limit your amps in the way i suggested,

let's take an example, 23A main motor curve, 17kmh field weakening start speed with a 1.5A variable, the scooter travels at 45kmh, aka its drawing 42 A of flux, aka Id

if we plop this into the formula, we get sqrt(23^2+42^2), which is 47.88~ A
there's an issue with that, that's on a 30A BMS limited system!
Iq changes throughout the speeds and draws way less than specified in the motor curve, resulting in the formula not being applicable, as in the lower speed curves this also flips around, with the Iq pulling more than the Id.

my proposal is the limiting of this using 1 singular value/slider with which the esc would on the fly perform this calculation to make sure it doesn't go above this limit.

paulederbaus, 2024-03-04:
> let's take an example, 23A main motor curve, 17kmh field weakening start speed with a 1.5A variable, the scooter travels at 45kmh, aka its drawing 42 A of flux, aka Id
> 
> if we plop this into the formula, we get sqrt(23^2+42^2), which is 47.88~ 

You are using the reference value and not the actual value. We have a current reading for both axes which can be lower or higher than what you request.

> my proposal is the limiting of this using 1 singular value/slider with which the esc would on the fly perform this calculation to make sure it doesn't go above this limit.
Are you aware that the prerelease has this option already as two sliders?

PureComedi, 2024-03-05:
Of course im using a reference value, because that's all you can set the 2 sliders to, fixed values

Nyami11, 2024-03-05:
> There is a very precise formula: I_total = sqrt(Iq² + Id²).

So we can limit the total current draw by setting the 2 values in a way that the I_total is the max that our BMS can push out, for example we want max 36A, which is 20A on Id and 30A on Iq, but this way until field weakeing turns on we are limiting it to 20A, which is much less than it could handle. If we would have a total power setting, it could work in a way that it lets more power to the d-axis until it starts field weakening, and after that it would reduce the current on the d axis in a way that the I_total would be under out total power limit. For example we could set Iq to 30A, Id to 36A, total to 36A, and then it would draw 36A until field weakening starts, then it would reduces the Id so sqrt(Iq² + Id²) is still less then our set limit.

closed 2024-10-09 by lekrsu