scooterhacking/scooterhacking-issues-public #46: BUG: Odometer frozen on SHFW on Ninebot E22

ahaas25, 2024-02-29:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

UI / UX fix

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

2.3.2

### SHFW Version

SHFW v0.3.6

### Scooter model

Ninebot E20, E25, E45

### Description of the Bug

Odometer does not update when using Nine Dash. Tested riding around 1 KM (through GPS) and then re-opened Nine Dash. Scooter reported exact same mileage. Reverting back to stock firmware allows the odometer to keep counting up, but does not account for miles ridden on the SHFW

### Steps to Reproduce

1. Flash latest SHFW
2. Ride scooter around at least 0.1 km
3. Check Ninebot Dashboard

### Expected vs. Actual Behavior

Expected: 500 KM on Odometer -> Ride 1 KM -> 501 KM on Odometer
Actual: 500 KM on Odometer -> Ride 1 KM -> 500 KM on Odometer

paulederbaus, 2024-03-01:
Could you please attach a screenshot of the value you are talking about?

ahaas25, 2024-03-01:
[image]
It's the same using both Nine Dash and Scootbatt

paulederbaus, 2024-03-04:
I think I found the bug. Are you willing to test the fix? If yes, please contact me via Discord.

ahaas25, 2024-03-04:
Contacted you on Discord.

closed 2024-09-19 by ahaas25