scooterhacking/scooterhacking-issues-public #50: Hall sensor reading range changes with headlight on/off

Nyami11, 2024-03-04:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

rc_support_shfw_037_v8_1

### SHFW Version

betapwm_Base187-v0.3.7_pre_v8_3

### Scooter model

Ninebot G30

### Description of the Bug

When you have the headlight on it reads a different value for the throttle and brake hall sensor. For me when I have the headlight off the throttle reads 40-41 when it's not pressed, and when fully pressed it reads 195. When I turn on the headlight it reads 41 on idle, and reads 180 when fully pressed.

https://github.com/scooterhacking/scooterhacking-issues-public/assets/116882110/db7fadb2-2997-4244-8afb-7d93301a661b

### Steps to Reproduce

1. Set the idle dash data to the throttle/brake value
2. Pull/press the throttle/brake and check what value it reads
3. Turn on the headlight and repeat step 2.
4. You shouls see a different reading.

### Expected vs. Actual Behavior

It should read the same hall sensor values with and without the headlights on

### Contact Information

DC: nyami11

paulederbaus, 2024-03-04:
This is a hardware issue existing in all firmwares. Can't be fixed with fw. - Issue closed.

closed 2024-03-04 by paulederbaus

nopbxlr, 2024-03-05:
Additional information: This issue arises from the 5V of the dashboard dropping with the load of the headlight. This ends up shifting down the max possible value of the hall sensor in the process. There is no clean way of fixing it from software, but the issue can be mitigated. Temporarily put "Brake value" or "Throttle value" as "Idle dash data" and turn the headlight on. Press the brake or the throttle fully according to the current selection and write down what the max value is. Use that as max values in the Hall sensor settings section. This will ensure you always get the full power/speed range from your inputs. You might have a small useless zone when the headlight is off, but nothing significant enough to be a bother.