scooterhacking/scooterhacking-issues-public #51: Auto Braking speed slider

JayTeel86, 2024-03-05:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Performance

### Priority

low (take your time)

### SHUT Version

2.3.2

### SHFW Version

0.3.6

### Scooter model

Ninebot G30

### Description of the new feature

A moving slider that allows the change of what speed auto braking can operate down until. i.e. Allow auto braking to be applied all the way down to 1km/h

### Contact Information

JayTeel1986

bodydisplaynone, 2024-04-23:
I +1 this for all models, where did this go? I remember having a speed setting (maybe I'm wrong) but now I only see current and not start speed.

oadq, 2025-06-13:
Autobrake now uses the brake minimum speed setting