scooterhacking/scooterhacking-issues-public #53: Feature: Negative Regen Braking

ArchGryphon9362, 2024-03-05:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Other

### Priority

low (take your time)

### SHUT Version

RC v3

### SHFW Version

0.3.7 Prerelease 5

### Scooter model

Ninebot G30

### Description of the new feature

I feel like we should be allowed to regen brake when travelling negative speeds. Perhaps an abs(speed) can be added somewhere in the check(s) whether or not to brake? This is nice when stopping at slopes or when trying to reorient the scooter (by holding brake to pull the scooter onto only one wheel). Thx!

### Contact Information

ArchGryphon9362

Nyami11, 2024-03-06:
It should also be able to fully lock up the motor when it stops braking backwards to help us re-orient it. This feature will only be useful on rwd scooters imo.

NormalHuman-Anything, 2024-03-07:
seems like very useful feature... I wanted negative speed brake lol

paulederbaus, 2024-06-20:
Allowing a negative minimum brake speed and changing the sign of the brake current seems too complicated for it's use case and the users who want it.
Perhaps the autohold feature can cover this.

closed 2024-06-20 by paulederbaus

LexNastin, 2024-06-20:
I meant more either
- take absolute value of min brake speed
- or only allow that when min brake speed is 0
and making it optional, rather than allowing changing the sign. but either way, autohold might be good enough, and there's enough other things to worry about for now. this is quite low priority, just a QoL change.