scooterhacking/scooterhacking-issues-public #54: BUG: Rename BLE xiaomi Pro

fox090594, 2024-03-12:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Other

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

2.3.2

### SHFW Version

0.3.7

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

after i changing the ble name, the name becomes to FLASH_NAME.
I create 2wd scooter. In 1s ble rename work okey. 

### Steps to Reproduce

1. Connect to ble
2. Change name
3. Reconnect

### Expected vs. Actual Behavior

Rename Ble name

### Contact Information

foxserzh

paulederbaus, 2025-02-10:
Works on my dev kit with BLE 1.2.9.

Do you have two original Dashes?

paulederbaus, 2025-02-11:
Closed until user feedback says otherwise

closed 2025-02-11 by paulederbaus