scooterhacking/scooterhacking-issues-public #58: Phase current on dash

lekrsu, 2024-04-10:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

rc_0-3-7-support-v8

### SHFW Version

0.3.7-RC1

### Scooter model

Ninebot G30

### Description of the Bug

Battery current with BMS emulation is said to have been fixed, but isn't. Both phase amps and wattage/100 on display isn't changing from showing 0. 

### Steps to Reproduce

1. Enable said options
2. Observe dash staying on 0 during current draw

### Expected vs. Actual Behavior

Expected dash to increase in value, didn't.

### Contact Information

lekrsu

lekrsu, 2024-04-10:
Somehow Scootbatt is able to report amps being drawn, assumably phase amps. This wasn't possible before.

NormalHuman-Anything, 2024-04-10:
yea I had same issue
scootbatt can read phase amps but no amps displayed(I run no comms and only bms emu)

https://github.com/scooterhacking/scooterhacking-issues-public/assets/72422431/dc3574f0-01d3-4fe3-857f-864b8d5bd656

closed 2025-06-23 by oadq