scooterhacking/scooterhacking-issues-public #60: Feature: Poweralgo (Delta)

Yawexi, 2024-04-12:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Performance

### Priority

high (definitely next release)

### SHUT Version

2.3.0

### SHFW Version

0.3.7

### Scooter model

Ninebot G30

### Description of the new feature

Hello, i want to suggest something. In the old cfw times there was a Parameter, named "Virtual Throttle Limit". Anyways this parameter does something else than what it is named like. It has surely something to do with the poweralgo. I hope that there is a way to recreate this parameter in the Beta, but you have to rename it. The parameter helps to run the delta mod. It makes the motor pull more current. 

### Contact Information

14S (Discord)

paulederbaus, 2024-04-12:
I don't see that parameter in a sh cfw generator. What generator did you use?

Was the default value 125, 128 or 183?

Yawexi, 2024-04-12:
It got removed from the CFW Tool of scooterhacking.

paulederbaus, 2024-04-16:
@nopbxlr Please help me. What was patched with this?

Yawexi, 2024-04-22:
I am generating my firmware through a Link because it is not possible to set the Parameter in the Firmware Generator of Scooterhacking. 
I just swap the last number of the link. Usually it is 120.

https://api.cfw.sh/max?output=zip2&version=DRV126&compat_patches=on&region=auto&remove_charging_mode=on&version_spoofing=on&showbatt=off&boot_workmode=3&voltage=36&p_sports=50000&p_drive=16000&i_sports=50000&i_drive=24000&i_eco=16000&max_speed_us=17&max_speed_eu=17&max_speed_de=17&direct_power_control=reg&direct_power_control_curve=quadratic&current_raising_coefficient=1000&motor_start_speed=1&brake_limit=130&brake_i_min=6000&brake_i_max=25000&kers_min_speed=5&no_kers=on&brake_current_raising_coefficient=500&brake_light_mode=static&brake_light_flash_frequency=0&cruise_control_delay=5&no_cruise_control=on&error_raising_level=3&no_overspeed_alarm=on&stay_on_locked=on&wheel_size=10.0&throttle_limit=97

paulederbaus, 2024-04-28:
You are using a bug of an older patcher version. If you increase your total current limit, you will have the same effect in SHFW.

closed 2024-04-28 by paulederbaus

Yawexi, 2024-04-28:
Sorry but it doesnt have the same effect. I tested it a few times already:
CFW 97 Coeff 25A set Phase Current -----> 36A Phase Current
SHFW 65A Phase Current -----> 50A Phase Current