scooterhacking/scooterhacking-issues-public #62: BUG:

sissj, 2024-04-19:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Crash

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.4

### SHFW Version

Unknown

### Scooter model

Ninebot G2

### Description of the Bug

Software refuses to connect to scooter, will appear in menu but will not connect, I have tried EVERYTHING! Was so excited fir this update but the app is still completely useless to me 

https://github.com/scooterhacking/scooterhacking-issues-public/assets/167530210/e7088a1f-9ae8-448e-a834-786df70c2480

[image]
[image]

### Steps to Reproduce

Use the app and try to connect to a G2 (EU)

### Expected vs. Actual Behavior

Dies not connect when prompted, will refuse and will crash on occasion 

### Contact Information

Discord: sissj        Telegram: +447494591623

BastelPichi, 2024-04-19:
Please note that this update only (is supposed to) fix connecting to the scooter (meaning SHU Support) but theres no SHFW support yet!

So dont get too excited just yet, you would only be able to see basic information.

sissj, 2024-04-19:
Hello mate yes I get youI'm referring to SHU app, I might not be missing out on a lot but it's quite annoying that I can't even connect the scooter to see what I could do with it, is a patch possible? Me and other users have chatted together and it's leaving us EU users hanging :( Thanks

nopbxlr, 2024-04-19:
Hello,
It appears that while we sorted out some bugs for the G2, another one appeared that is solely related to Android 14. We are currently looking into it, and are looking for someone with a G2 and an Android 14 device willing to do some debugging (mainly forwarding logcat data from a special build).
In the meantime, phones using older versions of Android should be able to connect properly.

sissj, 2024-04-19:
Ah yes I see, I'm on Andriod 14 so I'm happy to be a guinea pig if I am A:  guided through the process and B: it will not cause any harm to my G2 or my Andriod OS

nopbxlr, 2024-04-19:
This got fixed with the latest update (v2.4.1). Please update the app.

closed 2024-04-19 by nopbxlr