scooterhacking/scooterhacking-issues-public #63: Feature: Keep current performance mode for cruise control when switching modes

AGSPhoenix, 2024-04-19:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

UI / UX

### Priority

medium (upcoming releases)

### SHUT Version

SHUT 2.4 (68)

### SHFW Version

SHFW 0.3.6

### Scooter model

Ninebot G30

### Description of the new feature

## The problem
Switching between performance modes while cruise control is active can be dangerous, depending on the mode config and current speed.

Example steps:
Enable cruise control, and configure a slow Eco mode, and a moderately-faster Drive mode.
Accelerate up to full speed in Eco, and let go of the throttle to ensure cruise is active
Press the mode button to switch to Drive.

With one hand on the handlebars and one pressing the mode button, the scooter will lurch forward under the new performance mode. If this acceleration is strong enough, it will cause unintended steering towards whichever hand is still on the handlebar (and effectively pulling the handle back now that the scooter is moving forward faster than you).

Indeed, this happened to me, going from D to S at a healthy ~15-20 kph, and launching me into asphalt far faster than I could react, with very unpleasant results.

## Possible fixes
From a UX perspective, my suggested approach to this problem would be to accept the mode switch and only apply it as part of exiting cruise control. On other devices/vehicles, cruise means 'keep going this speed/power', but you can still mash the throttle for more power. This approach brings the same UX to SHFW's cruise mode.

Simply refusing the mode switch would also technically work, but I don't want to encourage an approach that restricts expert users who can make safe use of the current design. Smoothly transitioning to the new mode would also be reasonably safe, but figuring out how to fade between speed and DPC profiles sounds annoying to implement.

PureComedi, 2024-04-19:
I'm like 99% sure this was fixed already, there was a person who joined the dev team whose one main intention to fix this

AGSPhoenix, 2024-04-19:
I *just* flashed 3.7.0 and will try to verify whether this is still present soon.

AGSPhoenix, 2024-04-19:
I can conform this is still not fixed in 3.7.0.

paulederbaus, 2024-04-28:
Thanks for reporting this. Somehow I forgot to include this in the last release. 

New behaviour: When workmode is changed, cruise control is disengaged.

closed 2025-06-13 by oadq