scooterhacking/scooterhacking-issues-public #64: BUG: Updating to 0.3.7

PureComedi, 2024-04-20:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Performance issue

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.4.0

### SHFW Version

0.3.7 

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

The maximum field current setting or something else regarding field weakening gets messed up and either makes the scooter slower after updating from 0.3.6, or straight up makes it act as if field weakening is turned off

### Steps to Reproduce

Update to 0.3.7 from 0.3.6
Try to drive 

### Expected vs. Actual Behavior

Scooter go zoom vs scooter no go zoom

### Contact Information

purecomedi

SpookieGames, 2024-04-21:
Hey there, I have the same issue however resetting the settings doesn't help. Any ideas as to what to do next?

IIlIIllll, 2024-04-23:
i have the issue, that FW doesnt work after Shfw and shu update. i try tmrw to reset all shfw settings. hope thats the solution

paulederbaus, 2024-04-24:
This bug should be solved by updating to 3.7.1 - Please test and report back

IIlIIllll, 2024-04-27:
jup läuft wiederlgDaniel Kirchhoff

nopbxlr, 2024-04-27:
As this issue appears to be solved, it will be closed.

closed 2024-04-27 by nopbxlr