scooterhacking/scooterhacking-issues-public #65: BUG: Error 19 out of nowhere on esx and E25

avg-nikki, 2024-04-20:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

SHUT 2.4.1

### SHFW Version

SHFW 3.7.0

### Scooter model

Ninebot Es2, Es3, Es4

### Description of the Bug

(also present on E25) 
Errors 16 and 19 pop up with no reasoning behind them. Haven't seen this behavior myself on 0.3.6, 0.3.7 RC1, but can reproduce on release version. Multiple people are reporting issues of error 19 popping up after clean installing 3.7.0. 
[image]
[image]
[image]

### Steps to Reproduce

1. Have an esx scooter
2. Install 3.7.0

### Expected vs. Actual Behavior

Expected: No error.
Actual: Yes error.

### Contact Information

@kittyle on Discord, @el_khatto on Telegram. Will update this issue if I find out anything else.

avg-nikki, 2024-04-21:
thanks to @paulederbaus, seems to be fixed. will ride home today, update this issue, and we'll see how it goes.

[image]

avg-nikki, 2024-04-21:
According to a person in [🇷🇺 Ninebot | Segway chat & forum](https://t.me/ninebot_by_segway), installing base133 on esx drops error 16, and installing base255 drops error 19. Will check if I can reproduce that myself right now.

avg-nikki, 2024-04-21:
> According to a person in [🇷🇺 Ninebot | Segway chat & forum (Telegram)](https://t.me/ninebot_by_segway/38728), installing base133 on esx drops error 16, and installing base255 drops error 19. Will check if I can reproduce that myself right now.

[image]

avg-nikki, 2024-04-21:
Indeed, with 133, I get error 16, and with 255, I get error 19.
[image]
[image]

avg-nikki, 2024-04-23:
@paulederbaus managed to fix it up!  
[image]

NormalHuman-Anything, 2024-04-24:
update fixes it

avg-nikki, 2024-04-24:
> update fixes it

Sure does! 🎉 
Thanks, @paulederbaus!
[image]

closed 2024-04-24 by avg-nikki