scooterhacking/scooterhacking-issues-public #66: BUG:

SpookieGames, 2024-04-22:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.5

### SHFW Version

0.3.7

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

Scooter behaves as if the field weakening doesn't work. It uses less power than on 0.3.6 (by about 150W) and and the max speed on flat ground is like 33-34 while it was 40+ on the older version. I'd like to try to go back to 0.3.6 so that I can test it out it's not a hardware issue but a software issue.

### Steps to Reproduce

1. Have my scooter
2. Put in the setting from the guide
3. Scooter seems to not activate field weakening/not power it right

### Expected vs. Actual Behavior

Expected = scooter works fine (same as 0.3.6)
Actual behavior = scooter is slower than on the older version

### Contact Information

spookiegames

bodydisplaynone, 2024-04-23:
Did you try resetting ALL your shfw settings with? It should work after that.

SpookieGames, 2024-04-23:
> Did you try resetting ALL your shfw settings with? It should work after that.

It was the first thing I did. I'm gonna buy a new ESC and test it out on it.

VPSA132, 2024-04-23:
Hi. I have the same problem. FIELD WEAKENING does not work. After turning on this function, the scooter goes slower than without it.
In the application menu I only have the INITIAL CURRENT function. No VARIABLE CURRENT mA/km/h function as in the previous version. Application bug or intentional action?
[image]

bodydisplaynone, 2024-04-23:
Enable expert view. Reset ALL settings. Then configure everything again.

dee098, 2024-04-24:
top right corner

paulederbaus, 2024-04-24:
This bug should be solved by installing the update (3.7.1)

SpookieGames, 2024-04-24:
> This bug should be solved by installing the update (3.7.1) - Please try and report back

The scooter still uses 100-150W less energy than before and goes slower on flat surfaces (about 33-34km/h)

VPSA132, 2024-04-24:
"config -> top right corner" - how blind I am !!!
Now i can see variable current option.

paulederbaus, 2024-04-24:
The scooter still uses 100-150W less energy than before and goes slower on flat  ground (about 33-34km/h). Would there be a way to install the old 0.3.6 back so I can test out if it's a hardware issue or a software issue?

My dev kit is a Xiaomi Pro2 and field weakening works a 100%. Please revise your config or reset all settings.

paulederbaus, 2024-04-24:
> > This bug should be solved by installing the update (3.7.1) - Please try and report back
> 
> 
> 
> The scooter still uses 100-150W less energy than before and goes slower on flat  ground (about 33-34km/h). Would there be a way to install the old 0.3.6 back so I can test out if it's a hardware issue or a software issue?

My dev kit is a Xiaomi Pro2 and field weakening works a 100%. Please revise your config or reset all settings.

SpookieGames, 2024-04-24:
> My dev kit is a Xiaomi Pro2 and field weakening works a 100%. Please revise your config or reset all settings.

I already did that 7 times but still the same result. I will try different ESC tomorrow.

VPSA132, 2024-04-24:
Unfortunately, SHFW version 3.7.0 and 3.7.1 is no longer the same as 3.6.0. With the following settings, it is difficult to reach a speed of 35 km/h on 10" tires. I am lucky that I have an STM32F103 controller, I can upload the good old DRV155 which accelerates me to 37 km/h without field weakening. What is the problem? Is the new SHFW only for the controller v3.1 on GD32F103 MCU?
[image]
[image]

Nyami11, 2024-04-24:
@VPSA132 Did you tried setting the max fwk current to something higher?

VPSA132, 2024-04-24:
[Nyami11](https://github.com/Nyami11) NO. I tried to achieve the same results with identical settings as with SHFW 3.6.0.

VPSA132, 2024-04-24:
Ok, I checked what I could. Regardless of the Field Weakening settings, whether they are on max or min, the effect is always the same.  There is little power and it is difficult to accelerate to 35 km/h.  Scooter behaves as if Field Weaking was set to minimum current settings.  Any regulation doesn't work.  The Variable Current mA/km/h option does not work at all.
Please restore on app the verified working SHFW 0.3.6 (3.6.0).

NormalHuman-Anything, 2024-04-25:
btw max field current must be higher than max throttle A set or fwk will not have effect that much

NormalHuman-Anything, 2024-04-25:
aka crank that field current to 55a and torque to 65a

SpookieGames, 2024-04-25:
I can now finally confirm that the issue is not happening on version 0.3.6

SpookieGames, 2024-04-25:
> Ok, I checked what I could. Regardless of the Field Weakening settings, whether they are on max or min, the effect is always the same.  There is little power and it is difficult to accelerate to 35 [...]

Are you using a clone board by any chance?

VPSA132, 2024-04-26:
> Are you using a clone board by any chance?

I using original V1.4 STM32F103 with STP15810 MOSFETS and reinforced printed traces in the half-bridges.

Yesterday I set the following parameters on Sport mode:
45km/h max speed
26A current
Field weakening start speed: 25km/h
Initial current of field weakening: 3A
Variable current 1000mA/km/h
Max torque current 55A
Max field current 50A
10" tires

Result: Top speed 35km/h on flat surface. Low power, poor acceleration, you can't drive like that. 
The uphill speed drops to 21-23 km/h. 
On DRV155, the uphill (the same uphill of course a the same battery level) speed is 27-29 km/h.

In new version SHFW is another nice function:
Motor control frequency. This works but only affects the engine operating frequency before Field Weaeking is activated. It does not change the engine operating frequency after activating Field Weakening. It is hardwired to a value of 12 or higher.

VPSA132, 2024-04-26:
> just make sure throttle max a is under field current max

I'll test it today.

SpookieGames, 2024-04-26:
> > just make sure throttle max a is under field current max
> 
> I'll test it today.

Let me know the results

VPSA132, 2024-04-26:
Field current set to 55A and torque to 65A  -  no noticeable differences, same maximum speed.
Field current set to 65A and torque to 55A  -  no noticeable differences, same maximum speed.
I need to change ESC to v3.1 and see if anything changes.

SpookieGames, 2024-04-26:
> Field current set to 55A and torque to 65A  -  no noticeable differences, same maximum speed.
> Field current set to 65A and torque to 55A  -  no noticeable differences, same maximum speed.
> I need to change ESC to v3.1 and see if anything changes.

Is it an original v3.1 board? If so, could you share a link where to buy it?

VPSA132, 2024-04-26:
At the moment I have the original ESC v1.4 installed.
I will replace it with ESC v3.1 soon. Also original.
I have several. I bought them with damaged software from a person who works at an authorized service center. ST-LINK, 2 minutes of work and it works :).

SpookieGames, 2024-04-26:
> At the moment I have the original ESC v1.4 installed.
> I will replace it with ESC v3.1 soon. Also original.
> I have several. I bought them with damaged software from a person who works at an authorized service center. ST-LINK, 2 minutes of work and it works :).

Nice 👍

SpookieGames, 2024-04-26:
I will try to install an original 2.1 ESC and I hope that works fine

SpookieGames, 2024-04-26:
> Theoretically, after changing the driver, nothing should change for the better. The half-bridge control circuits and current measurement circuits are the same in v1.4, 2.1 and v3.1. The only differences are the type of microcontroller and different (weaker in version v3.1) transistors in the half-bridges. This has no effect on the operation and power of the engine.

But does the theory actually work in the reality?

VPSA132, 2024-04-26:
Well, if you don't test it experimentally, you won't know. 
Luckily, I have one ESC with an old version of SHFW (3.6.0). However, there is a problem. The SHU 2.5 application is not compatible with the old SHFW 3.6.0. You need to download the old version of the SHU 2.4.1 application and run it without internet access. Then you can safely change the parameters.

SpookieGames, 2024-04-26:
So, I've changed the v3.0 clone ESC I had for an original v2.1 ESC and 3.7.1 works the way it should. I'd even say that the scooter is a lot faster.

lekrsu, 2024-04-26:
> So, I've changed the v3.0 clone ESC I had for an original v2.1 ESC and 3.7.1 works the way it should. I'd even say that the scooter is a lot faster.

Good that the original hardware works as intended now

SpookieGames, 2024-04-26:
> > So, I've changed the v3.0 clone ESC I had for an original v2.1 ESC and 3.7.1 works the way it should. I'd even say that the scooter is a lot faster.
> 
> Good that the original hardware works as intended now 

Yeah but that doesn't explain that VSPA is having the same problem with original hardware

VPSA132, 2024-04-26:
Great info! The new SHFW is not suitable for the old ESC v1.4 drivers. Just replace the ESC and the problem is over.
Can you share yours settings here?

SpookieGames, 2024-04-26:
Sports - 25A flat curve
Drive - 20A flat curve
ECO - 15A flat curve
[image]
[image]

SpookieGames, 2024-04-28:
As I have used an original ESC and everything works I am now going to close this issue

nopbxlr, 2024-04-28:
Thanks for confirming the issue happens with clone ESCs! However, it also seems to happen to v1.4 ESCs. The issue must be kept open as there seems to be a bug related to hardware differences

VPSA132, 2024-04-30:
I confirm. On ESC v2.1 SHU3.7.1 works very well! Appropriate performance and a maximum speed of 40 km/h with good settings are achieved without any problems. 35km/h on an empty battery (25%) :D on flat terrain.
You can feel the moment of FIELD WEAKENING turning on if you set the Initial Current to a value other than 0A. Interesting fact - my ESC v2.1 has an STM32F103**C8T6** MCU.
v3.1 is waiting for testing but I think it will be ok.

paulederbaus, 2024-05-08:
Spoke confirmed that new hotfix works with clone esc. I labelled it with next release and hopefully we can close it then.

paulederbaus, 2024-06-20:
Please confirm this is closed with 3.7.2.

SpookieGames, 2024-06-20:
> Please confirm this is closed with 3.7.2.

Sadly I cannot as I no longer have the clone ESC but it worked fine in the Regular-Base247-v3.7.1_fwk_hotfix_v3 version.

paulederbaus, 2024-09-19:
Issue should be solved with 3.9.x

closed 2024-09-19 by paulederbaus