scooterhacking/scooterhacking-issues-public #67: BUG: ReFlasher 1.4 bricks F series controllers

bodydisplaynone, 2024-04-23:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

critical (app crash, data loss)

### SHUT Version

2.5

### SHFW Version

0.3.6

### Scooter model

Ninebot G30

### Description of the Bug

**_Can't select it from the model list but this happened to an F40E._** This is something more users experienced as of now according to the discord channel. Some versions of the reflasher might soft-brick the F series esc after stlinking resulting in a communication error between the BLE and the ESC.

### Steps to Reproduce

1. Stlinked controller with the latest ReFlasher 1.4
2. Console showed done, all ok.
3. When plugged in the scooter turns on but immediately beeps and blinks red Error 10

### Expected vs. Actual Behavior

The expected result was a working controller. The actual result was a soft-bricked one and no matter what I did I couldn't make it work, I even tried to connect SHU but it got stuck on 'pinging scooter'...

The fix was to use an older version, I used the 1.2 based on a YouTube tutorial and it worked, after plugging in the ESC I got a working scooter immediately.

### Contact Information

@sealhe

nopbxlr, 2024-04-27:
The issue has been identified. After writing the bootloader, it writes the BLE firmware instead of the DRV firmware due to a bug in the code. A fix is been implemented and a release should follow today.

nopbxlr, 2024-04-27:
ReFlasher was updated to v1.4.2 which solves this issue. Thanks for the report, and don't forget to update the app!

closed 2024-04-27 by nopbxlr