scooterhacking/scooterhacking-issues-public #68: BUG: Motor is turning off (?) when accelerator is not pressed on F series.

bodydisplaynone, 2024-04-24:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.5

### SHFW Version

0.3.6

### Scooter model

Ninebot Es2, Es3, Es4

### Description of the Bug

**Confirmed on F series but has no select option in the scooter model field.**

Presumably, the motor turns off when the throttle is at idle, causing a very erratic and possibly dangerous behaviour as the motor turns on/off each time the accelerator is pressed and the whole scooter pushes-pulls-shakes very aggressively.

This is just a guess but other users also confirmed this behaviour on an F30, mine is an F40. The 'switch of motor' toggle doesn't seem to affect this.

https://github.com/scooterhacking/scooterhacking-issues-public/assets/22660372/31b1782a-8e7c-4da4-95f1-0ce6be80f62b

### Steps to Reproduce

Installed shfw 0.3.6 and set it up as usual. Tried both speed-based and dpc power, no change in behaviour.

### Expected vs. Actual Behavior

I think the motor should not kick every time the throttle is pressed.

### Contact Information

@sealhe

paulederbaus, 2024-04-24:
What F-Series is this and what SN did you use?

bodydisplaynone, 2024-04-24:
> What F-Series is this and what SN did you use?

@paulederbaus This is an F40E from 2022, current serial is N5GED0000C0000 set by ReFlasher 1.2. The only reason I used 1.2 is because of this issue:
https://github.com/scooterhacking/scooterhacking-issues-public/issues/67

I reverted to base 5.4.9 from SHU with this serial and the throttle/motor works normally.

paulederbaus, 2024-04-28:
Have you used the switch in system settings that switches off the motor on throttle release? (Please provide your system settings as a screenshot)

bodydisplaynone, 2024-04-29:
I don't have access to the scooter at the moment as my girlfriend is using it but yes, I used the switch and tested both states.
By default, the function is deactivated (so the motor is supposed to stay on), still causing the behaviour seen in the video.

After that, I tried switching this function on and off but it did not make a difference.

bodydisplaynone, 2024-05-23:
@paulederbaus I managed to install shfw and try again, switching the option on and off but there is no difference.

https://github.com/scooterhacking/scooterhacking-issues-public/assets/22660372/389ab9c4-7c54-4cd3-9766-66d41c89c134