scooterhacking/scooterhacking-issues-public #69: BUG: Jerky braking

VladimirKarpunin, 2024-04-25:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.5

### SHFW Version

3.7.1

### Scooter model

Ninebot E20, E25, E45

### Description of the Bug

Problem when braking to 0 km/h on E25, ESX (clonk, clonk). When it almost stops completely, clicks are heard from the motor. There are no problems on two similar scooters with version 0.3.6 and stock firmware. Other users also have this problem.
All sensor hall checked.
This problem only occurs when the brake is applied at low speed.
There are no problems with accelerating and braking. The problem is only with a full stop.
[image]
[image]
[image]
Video with problem:
[https://youtu.be/c8kTgDdO70E?si=j54DcqFhcDnZIOSS](url)

### Steps to Reproduce

1. Download SHU 2.5
2. Flash firmware 3.7.1
3. Configuration Brake start 0kmh

### Expected vs. Actual Behavior

Expected result:
Smooth braking as on previous versions of SHFW.
Actual result:
Stopping when braking occurs jerkily.

avg-nikki, 2024-04-26:
Probably has something to do with the fact that brake boost was introduced. (possibly, something was changed in the braking algorithm?). Anyway, +1 this, because I'm experiencing the same on an ES4: the motor is trying to stop really abruptly at lower speeds, and starts "clicking". Also, braking and trying to roll it shows this issue pretty well.

paulederbaus, 2024-06-20:
Please recheck this with upcoming 3.8.0 release.

closed 2024-09-19 by paulederbaus