scooterhacking/scooterhacking-issues-public #74: adding sensorless control

NormalHuman-Anything, 2024-05-02:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

UI / UX

### Priority

medium (upcoming releases)

### SHUT Version

2.5

### SHFW Version

3.7.1

### Scooter model

Ninebot G30

### Description of the new feature

sensorless control (in case like motor hall senor escapes its original position so it won't work , running some different motor with 120° hall sensor angle or etc)
would be a very good feature to have. had to walk home 10km because hall sensor decided to escape the position Caz glue got loose.
and also option to reverse the direction if it's spinning backwards or etc

### Contact Information

Jason on discord

NormalHuman-Anything, 2024-05-02:
[image]

paulederbaus, 2024-05-08:
This feature requires too many code changes to be implemented in a regular DRV. We know how different F-Series DRV is. Sorry but we won't be able to implement this.

closed 2024-05-08 by paulederbaus