scooterhacking/scooterhacking-issues-public #75: BUG: PWM frequency not applied to Field Weakening

Z1mmb0, 2024-05-06:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

latest greatest

### SHFW Version

3.7.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

When FW starts, the PWM frequency seems to go up. Custom settings are ignored when FW hits.

Maximum speed is compromised.

### Steps to Reproduce

Xiaomi pro2.

I_d limit: 35 (default)
Iq limit: 65

FW settings: 
start 25km/h
init: 2A
variable: 1200 mA

go 25 km/h, different noise characteristic audible at this point.
go to max speed: top speed lowered in comparison to previous shfw version.

### Expected vs. Actual Behavior

expected: 
resulting performance with same parameters compared to 0.3.6

actual:
scooter goes slower, needs more agressive settings, possible risk of esc overheating by accident

### Contact Information

z1mmb0

Nyami11, 2024-05-06:
It changes the PWM frequenczy to 16kHz with field weakening, if it's under it for safety reasons. At lower PWMs there was a strong pushback when releaseing the throttle at high speeds (especially at 4kHz), that's why the automatic changes was implemented. PWM frequenczy shouldn't effect the top speed/torque at high speeds

paulederbaus, 2024-05-06:
If you are willing to test something, that would be great. Then please write me your uid in discord or telegram.

paulederbaus, 2024-05-08:
possible hot fix with 3.7.2 - still waiting for your feedback

oaddyk, 2024-06-08:
I'm also experiencing this issue. I updated from 0.3.5 and my speed has dropped from around 42km/h to 34km/h. Before it was drawing around 750W, now it's drawing only 600W.

I am using the same exact settings as before.
Start speed 20km/h, Initial 0A, Variable 1.3A
Increasing the max field current did not make a difference, and if it did it doesn't really matter, it's still an issue.

I would be willing to test the hot fix if possible.

0.3.5:
[image]
3.7.1:
[image]

I should probably make a GitHub Issue about this, but the Brake boost not working as intended, at least from what I've seen people say, on FWD scooters like my Pro 2 around 30% should be enough to make a noticeable difference, I did not feel any difference, the current did go up from -6A to -9A but there was no difference feeling wise, I'm going to try 70% on Monday.
One more thing, this might not be true at all, but it feels like acceleration after around 20km/h is just weaker than on 0.3.5, I'm not sure about this it's just a feeling.

oaddyk, 2024-06-10:
> I'm also experiencing this issue. I updated from 0.3.5 and my speed has dropped from around 42km/h to 34km/h. Before it was drawing around 750W, now it's drawing only 600W.
> 
> I am using the same exact settings as before. Start speed 20km/h, Initial 0A, Variable 1.3A Increasing the max field current did not make a difference, and even if it did, it doesn't really matter, it's still an issue.
> 
> I would be willing to test the hot fix if possible.
> 
> 0.3.5: ![image](https://private-user-images.githubusercontent.com/48562304/337925797-f6c6d050-e13f-4b1d-bfe5-fd0f864ee105.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgwMjg2MjEsIm5iZiI6MTcxODAyODMyMSwicGF0aCI6Ii80ODU2MjMwNC8zMzc5MjU3OTctZjZjNmQwNTAtZTEzZi00YjFkLWJmZTUtZmQwZjg2NGVlMTA1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEwVDE0MDUyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0NDMyZjUzMGVlY2ZhZDU4MmQxZTI5MGJkMWUxYjdjZjIxMWVjM2QzYjY2ZDQ2NzQ3NWU3YWRjMmY1OTAxODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rc2nWppo4nHiwW1LbYOQFafDuRZLuKGSaStsD2xCbVg) 3.7.1: ![image](https://private-user-images.githubusercontent.com/48562304/337925799-064e4ebd-b1c4-4e56-8ad4-b1a09a543713.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgwMjg2MjEsIm5iZiI6MTcxODAyODMyMSwicGF0aCI6Ii80ODU2MjMwNC8zMzc5MjU3OTktMDY0ZTRlYmQtYjFjNC00ZTU2LThhZDQtYjFhMDlhNTQzNzEzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEwVDE0MDUyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNiMTAzOWViM2UxMDMzZDljNzNkZDVkOWRkN2RmY2MyMDA1ODEyYWFhN2MyYzc2YzdjNjI4YTQzYTgzNTg3OTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yHeolPTDm5xNBh9fUk-Vf_oP733GgOjxQhg7hciOL6o)
> 
> I should probably make a GitHub Issue about this, but the Brake boost not working as intended, at least from what I've seen people say, on FWD scooters like my Pro 2 around 30% should be enough to make a noticeable difference, I did not feel any difference, the current did go up from -6A to -9A but there was no difference feeling wise, I'm going to try 70% on Monday. One more thing, this might not be true at all, but it feels like acceleration after around 20km/h is just weaker than on 0.3.5, I'm not sure about this it's just a feeling.

Tried 70% brake boost, 0 difference, there's no difference, I'd say the e-brake was stronger on 0.3.5.

paulederbaus, 2024-06-20:
Have you checked this with 3.7.2? Please also increase the brake overshoot to 65A for testing purposes.

oaddyk, 2024-06-21:
> Have you checked this with 3.7.2? Please also increase the brake overshoot to 65A for testing purposes.

hey, we talked on discord in the #shfw channel about this, but ill give you an another update
the max speed is not as high as it used to be (42-45kmh), i can only reach (39-40kmh)
downhill the speed is the same as it used to be on 0.3.5, which is different than what other people reported (around 60kmh) going downhill with a pro 2, i tried 2 different FW settings, made no difference.

the brakes are stronger, but not as strong as theyre supposed to be
30A curve, 65A overshoot, and brake boost at 100% should flip me over, which it does not, but at least its stronger than it used to be on 0.3.5

Z1mmb0, 2024-06-23:
at this point 0.3.6 is too much in the past. it needs some data base with dedicated test to track the performance. tbh the difference margin is quite low. subjective feelings will not lead to anything long term. i suggest there should be a requirement for performance reports.

anyway.
1st some comment towards the original report. 
In the meantime i somewhat managed to reach the previously achieved speeds, by lowering the torque and applying more FW and 100% accel boost instead. in terms of pwm freq i just went to 24khz. but ofc it costs the fun factor which was introduced with accel boost. 
scooter goes currently 40-41 vs 42 kph. (@48V). pretty much torque limited at those speeds above 36 kph.

the hotfix i tried was pretty crazy. i got so much wheelspin that it was dangerous to drive with. but with nerfed torque value parameters it might get driveable.

new topic break: 
break is set to 31A (linear) with overshoot to 65A. Break boost 100%. lowest break speed is set to 3kph. from my pov the break works fine enough.
but i experience some motor rippling in lower speeds while breaking (between 3-10 kph). happens in a inconsistent manner.

oaddyk, 2024-06-23:
> at this point 0.3.6 is too much in the past. it needs some data base with dedicated test to track the performance. tbh the difference margin is quite low. subjective feelings will not lead to [...]

i updated from 0.3.5 like 2 weeks ago so not really

Z1mmb0, 2024-09-09:
shfw 3.9.1:

fPWM got changes and now works as intended. While Fwk is active, shfw doesnt bump up the pwm freq to 16khz+ anymore. smoke tested with 8,12 and 16khz. 

closing as original topic is done.

closed 2024-09-09 by Z1mmb0