scooterhacking/scooterhacking-issues-public #77: Feature: checkbox for flashing shfw while using a display from a different scooter (ninebot)

Kebabbabek, 2024-05-15:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Communication

### Priority

low (take your time)

### SHUT Version

2.5

### SHFW Version

3.7.1

### Scooter model

Ninebot G30

### Description of the new feature

I have got an esx display connected to my g30 controller, and i think it should be easy to add a checkbox stating you're using a different ble, and you can enter your actually controller/scooter model. I believe this works with most ninebot scooters, as they use the same ble protocol. This could prevent people from bricking their scooters and helping people that don't have the right parts.

### Contact Information

Fog0327

Kebabbabek, 2024-05-15:
Sorry for my typos

paulederbaus, 2024-05-22:
The scooter is not identified by the DRV. This is all done by BLE. You will need to change the beacon in the esx ble file to do this.

closed 2024-05-22 by paulederbaus