scooterhacking/scooterhacking-issues-public #78: Feature: Advanced Poweralgo

Yawexi, 2024-05-15:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Performance

### Priority

low (take your time)

### SHUT Version

2.3.0

### SHFW Version

0.3.7

### Scooter model

Ninebot G30

### Description of the new feature

More Expert Options, to optimize the power algo of the motor. (useful if you dont use the stock motor or use delta motor)

### Contact Information

0.moon_light.

IAMFreehandZ, 2024-05-15:
I think the motor algo in shfw is pretty much hard coded, so I think it'd be very time consuming for Paul to write a new algo for a very small group of people.

paulederbaus, 2024-05-22:
Please explain what you have in mind and how it would change the scooter's behavior.

closed 2024-05-24 by Yawexi