scooterhacking/scooterhacking-issues-public #80: BUG: G30 error when charging and using ninedash

SET19724, 2024-05-19:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

not used in this case

### SHFW Version

3.7.1

### Scooter model

Ninebot G30

### Description of the Bug

When I was charging the G30 Max, I used Ninedash to check the battery level. After a few refreshes, the meter remained at, for example, 51%. Only after disconnecting the power and holding the button for a while, it initially did not respond to the button and indicated that it was still charging for 1-2 minutes. After that time and holding the button, it finally turned off, and after turning it on again, it showed 77%.

### Steps to Reproduce

1.Charge
2.Use ninebot app
3.Check battery status
4.Also use update data in notification when it display charging
5.It get that issue

### Expected vs. Actual Behavior

Now, after the restart, it works, but it shouldn't freeze.

Z1mmb0, 2024-05-24:
experiencing a related issue with pro2 and 48V battery. Not sure what happened, but charging is stuck on 77%. Tried to disconnect the battery. Doesnt heal the issue for me.

paulederbaus, 2024-09-19:
Is this still present?

closed 2025-02-10 by paulederbaus