scooterhacking/scooterhacking-issues-public #83: Feature:

mxstiii, 2024-05-25:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Other

### Priority

high (definitely next release)

### SHUT Version

X

### SHFW Version

X

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

I wanted to ask if yall adding the Xiaomi 4 Series like the 4 lite 2gen i would like to havk my scooter too

### Contact Information

Mxstiii

lekrsu, 2024-05-25:
Low effort posts like this won't get you far.

Nyami11, 2024-05-25:
Unlike older xiaomi models which were made by Ninebot, it's made by a totally different company. SHFW and other SH tools are designed for ninebot made scooters, since the firmware and communication protocols are pretty simular on all of them. Making the tools compatible with that model would require a lot of work.

BastelPichi, 2024-05-25:
> Unlike older xiaomi models which were made by Ninebot, it's made by a totally different company. SHFW and other SH tools are designed for ninebot made scooters, since the firmware and communication [...]
And even if, they would add Brightway before LEQI.

And why didnt you just ask on the discord? We would have given you this exact answer directly...

paulederbaus, 2024-09-23:
Some of the new Xiaomi models are not manufactured by Ninebot like previous models, but by Brightway (Navee).
These models are not supported by SHU/SHFW, and if they will be is uncertain. The following models are affected:

- Mi 3 Lite
- Mi 4
- Mi 4 Ultra
- Mi 4 Lite
- Mi 4 Pro 2nd Gen (has a different controller, not supported with the tutorial below)

If you feel adventurous and have a model limited to 20 km/h (e.g. Germany), it's possible to change the serial number to the global region with an ST-Link (thanks to dnandha/Turbojeet!):

closed 2024-09-23 by paulederbaus