scooterhacking/scooterhacking-issues-public #85: No brake light response when braking, ghost throttle, and severe vibration when field weakening

NormalHuman-Anything, 2024-05-31:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

g2 rc v1

### SHFW Version

3.8.0 g2 RC

### Scooter model

Ninebot Es2, Es3, Es4

### Description of the Bug

brake light doesn't blink when braking 
when field weakening it has ghost throttle and severe vibration with limited freewheel speed(60 > 45 ish)

### Steps to Reproduce

install 3.8.0
test

### Expected vs. Actual Behavior

expected : same as 3.7.0
actual : above mentioned

### Contact Information

Jason

Nyami11, 2024-05-31:
Field weakenig doesn't seem to work. On G30, I wasn't able to hit 30, instead of the 40 that I was able to do earlier. It just vibrates like crazy. I uploaded video on prerelease channel about the vibration.

There is also an issue where if I lightly tap the throttle, it still gives power after I release it for a short time.

NormalHuman-Anything, 2024-05-31:
that's exactly same behavior as mine

Nyami11, 2024-05-31:
Thost throttle is present even without fwk. 
Idk idk if brake light issue is present on G30, since my light is broken.

NormalHuman-Anything, 2024-05-31:
https://discord.com/channels/550154171433615364/1107741792738222091/1246016621466746921
btw the video

paulederbaus, 2024-06-20:
Fixed and confirmed by users.

closed 2024-06-20 by paulederbaus