scooterhacking/scooterhacking-issues-public #87: Feature: Put scooter in lock state on startup

arths31, 2024-06-07:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Other

### Priority

low (take your time)

### SHUT Version

None

### SHFW Version

SHFW 3.7.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

Hello,
To protect my scooter a little bit more, I’m using a profile on startup with no speed curve and maximum breaking.
Using throttle and break combination, I then switch to the normal profile when I want to drive.

Would it be possible to implement a feature where the scooter starts directly into lock mode instead and could be then unlocked with a combination set by user ? (just like profile switching)

Keep up the good work
Thank you

NormalHuman-Anything, 2024-06-08:
just use profile switching

arths31, 2024-06-08:
I do use a U-Lock as well

arths31, 2024-06-08:
Just had an idea for easier implementation.
A switch could be added when you are configuring a profile that sets it as a « lock profile » thus triggering the lock state.

This way we can trigger it like any other profile with profile switching and/or set it on startup.

I can beta test on my scooter if needed.
Cheers

PureComedi, 2024-06-08:
such features are basically useless it would only take up valuable space and serve no extra purpose as it's just like 4 taps to set a profile to start on boot and have no power

Nyami11, 2024-06-08:
Depending on how profile triggers are implemented, it might be easy to run the scooter locking function on a trigger, instead of the profile switching.

arths31, 2024-09-02:
@nopbxlr I guess we can close this issue as it seems related to #102 😄