scooterhacking/scooterhacking-issues-public #90: BUG: doesn't connect to connect to Xiaomi mi pro4

Nakis122, 2024-06-11:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Other

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

Don't know

### SHFW Version

Don't know

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

It doesn't connect to Xiaomi mi 4pro

### Steps to Reproduce

Just doesn't connect/Bluetooth doesn't connect 

### Expected vs. Actual Behavior

No results 

### Contact Information

nak_is  for discord

lekrsu, 2024-06-11:
4 Pro is not supported nor is it in the supported model list when you open the app.

Nakis122, 2024-06-11:
> 4 Pro is not supported nor is it in the supported model list when you open the app.

do you know any other apps that work with the mi 4 pro?

Nyami11, 2024-06-11:
> do you know any other apps that work with the mi 4 pro?
>

There are no public tools for it, and probably never will be.
Close this issue please. If you have other questions, ask on the SH discord server

Nakis122, 2024-06-11:
okay

closed 2024-06-11 by Nakis122