scooterhacking/scooterhacking-issues-public #91: Feature: power (10W) dash data

Xc987, 2024-06-14:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

UI / UX

### Priority

low (take your time)

### SHUT Version

2.5

### SHFW Version

3.7.2

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

New main / alternating dash data: Power (10W).
When the power is 300, the dashboard should show 30.
Useful for scooters that have motor power bellow 1000W.
It's not meant to replace the Power (100W), but as an additional dash data for lower powered scooters where more precise data would be useful when riding.

paulederbaus, 2024-06-20:
I think there will be more users that have a power consumption between 10W and 1.690W. The logic of this display will be changed with the next update.
---
Issue will be closed with release.

Xc987, 2024-09-12:
Feature added in 3.9.0, closing issue.