scooterhacking/scooterhacking-issues-public #92: Feature:

PureComedi, 2024-06-16:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Communication

### Priority

low (take your time)

### SHUT Version

.

### SHFW Version

.

### Scooter model

Ninebot G30

### Description of the new feature

Gonna be real conclusive,

# **Remove expert view**

It so far served absolutely 0 purpose other than being 1 more thing that users have been getting caught on, it is nothing but an extra, useless step.

### Contact Information

purecomedi

Nyami11, 2024-06-16:
Yep, I don't see the point of having it too. In my opinion, it should be implemented in a way that would make the settings simpler for new users in a way that wouldn't compromise performance. One better way of implementing would be in my opinion is if it would merge some settings into one, which would adjust mutliple settings at once to recommended values. For example field weakening without expert mode could be one "fwk level" slider, which would adjust all 4 settings. For example level 1 would set a 25 kmph start speed, 300 variable, 0 initial, level 5 would set 22 start speed 1000 variable, 3 initial. Motor PWM slider could be turned into a "quiet motor" switch that would switch, which would set it to 8kHz or 20kHz. Just like these, other settings could be simplified and that would make non expert mode usefull for new users, but honestly, reworking it seems like a waste of time.

lekrsu, 2024-06-16:
The idea of decreasing the exposed options for new users, _who might even miss that there's an option to view more_, is counter intuitive. I might be bold to ask this, but a complete config menu re-design would be useful. Right now the menus are sub menus within sub menus. Would love to work on it if we had source access, at least for the UI part.

paulederbaus, 2024-06-20:
I will kill myself... 🤣 
If it's easier for you to support users, I can try to enable it by default.

> One better way of implementing would be in my opinion is if it would merge some settings into one, which would adjust mutliple settings at once to recommended values. For example field weakening [...]

We have a strong "no spoon-feeding" policy. The settings will stay as precise and "complicated" as they are. We will not make automatic settings that set a different state than what Ninebot or Xiaomi intended. If you want a faster scooter, understand what field weakening is. If you want a quieter motor, understand what's happening with changing the motor's PWM frequency.

Nyami11, 2024-06-20:
I just gave my idea on how it could me made it usefull for new users. Honestly, I don't think we should have an expert mode setting at all. If a user installs SHFW, he should do research. Hiding settings to protect them is kinda useless imo.

Z1mmb0, 2025-04-02:
tbh its just the positioning of the expert toggle switch. easy to overlook. 
Most users dont really dial in their settings with care - its good to initially hide the more risky features behind a "im a advanced user" switch. 
Otherwise you get a lot of "i broke my scooter" questions. The all the amps meme doesnt come from nowhere... 
Knowing every possible implication of some settings might take several hours of work. I suggest interpreting the expert mode as a learning roadmap, not as a user protection.