scooterhacking/scooterhacking-issues-public #93: Feature: Multiple SHFW versions available to install

taubensuppe, 2024-06-16:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Communication

### Priority

medium (upcoming releases)

### SHUT Version

2.5

### SHFW Version

3.7.2

### Scooter model

Ninebot G30

### Description of the new feature

It would be a great addition if users could manually select between public SHFW's themselves, depending on their liking and which SHFW version they found best (in terms of Bugs, Throttle Response, Efficiency, Features, Motor / fwk algorithms).

I am using SHFW since February 2023, and ofcourse I was always excited about new features and presumably “better” SHFW versions. I personally stopped updating SHFW when I was at 0.3.5, I was fine, everything worked very stable, efficient and the features it came with were enough for me. I tried out the 1.8.7 DRV Base 2 times, but always went back to the 173 Base, as long as it was possible, which sadly isn’t the case now anymore. 
I understand that recommending users to update and only show the very newest and single SHFW version is meant to fix bugs of previous SHFW versions, as well as add new features one could like. I’m just not a big fan of the way it’s implemented, I would really like to have a way to install older SHFWs onto my scooter which I personally just found better overall. Maybe add a button in the SHU SHFW installation prompt that lets you also choose from older versions, thus informing the user that the current config or some features will be lost if he downgrades.

For example, since I updated SHFW last week I had a bigger battery drainage without changing any of my settings. Even if acceleration and brake boost is a great feature, I gotta say that I just can’t get my scooter to the same state / behavior like it was on DRV173, speaking in terms of vmax, climbing up hills etc. In my opinion this newer version adds great new features, but also has its downsides, which older SHFWs may didn’t have.

Why not let the user decide which SHFW version he wants to install on his scooter, depending on his needs and personal liking?

### Contact Information

varterde

paulederbaus, 2024-06-20:
I understand that there may be some bugs in newer versions of SHFW that need to be fixed. The best way to get rid of them is to help us fix them.
Your observations about 1.7.3 and 1.8.7 are strange and incomprehensible to me, since the only difference between these two versions is the PWM frequency.

None of this is really relevant to the availability of older versions of SHFW. These versions will lead to bugs with SHUT because we are constantly changing the interaction between the app and the FW.

lucienne3-web, 2024-07-01:
this should be a good feature to implement !! why not ??

nopbxlr, 2024-07-02:
Unfortunately, as explained above, keeping compatibility regardless of the Utility/SHFW version combo is too big of a task and we spend all of our time already making new and improved updates, as well as fixing bugs

closed 2024-07-02 by nopbxlr