scooterhacking/scooterhacking-issues-public #99: BUG: External Battery not recognized in Ninebot E series

lucienne3-web, 2024-07-01:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.5

### SHFW Version

3.7.2

### Scooter model

Ninebot E20, E25, E45

### Description of the Bug

External battery not recognized in E series . Attached screenshoot of internal and external battery status at the moment !
it is also impossible to up/down grade external battery BMS using SHU utility ( because is not recognized , i think ) 
Test made : Return to OFW 2.7.1 and connected the charger , the external battery start working again !

### Steps to Reproduce

to reproduce the issue simply install SHFW on E45 , starting from earlier 2022 version and update everytime a new update was available . 

### Expected vs. Actual Behavior

have 20 km of autonomy instead of 10

### Contact Information

lucienne3

NormalHuman-Anything, 2024-07-03:
not me but also heard that from other es4 user i know
somethings gotta be wrong...

LarsMde, 2024-07-22:
I can confirm the behaviour above.

ESC 2.6.11 - BLE 2.1.3 - BMS 1.6.0 - eBMS 2.4.1

After the upgrade, I first fully charged the entire system (A0). On the first ride, after a few kilometres, I noticed that the battery was discharging unusually quickly (when I checked with the ScootBatt, however, it showed that only the internal battery was discharged, the external (original) battery was still at 80%). What was particularly unusual was that the 4 white LEDs on the additional battery were all flashing permanently at the same time. This also happened immediately after switching the entire system off and on again. Otherwise, there were no error messages within the ScootBatt or defective cells.
If I connect the scooter to the charger in this state, it charges both batteries (first the internal one and later, when both are at approximately the same level, the others as well). However, the LED on the charger keeps jumping from red to green, red to green, etc. (the green phase is always slightly longer). (the green phase is always a little longer) and the scooter beeps again and again as if restarting (the front headlight flashes briefly at the same time). After a few minutes this normalises and it charges without flashing, beeping and green phases on the power pack.
I reproduced this behaviour yesterday and today. After the scooter is fully charged, it initially behaves normally when riding. After approx. 1 km, the 4 white LEDs of the additional battery start to flash rhythmically. After travelling approx. 12 km, the battery indicator shows 40%.
It is also noticeable that the ScootBatt app shows me the serial number and status of both batteries, but the firmware of the eBMS is 0.0.0 (I know from a screenshot from 2 days ago that it said 2.4.1). So there seems to be basic communication with the additional battery.

I have now flashed back to 2.7.1 (compat). Now the eBMS is also displayed correctly again (2.4.1) and the scooter is behaving normally (both batteries discharged in parallel - range approx. 40km).

The problem is therefore related to SHFW.

lucienne3-web, 2024-08-02:
the SHFW versions released last year doesn't have this issue ... why we can't have it ??

LarsMde, 2024-08-02:
If this is a security feature from the developers' point of view, that the versions are not freely downloadable, then I would first ask that in the future, as with the firmware of DRV, BMS or BLE, at least the selection of the latest version and the version from last year, in which the additional battery still worked, is offered.

Thanks.

BastelPichi, 2024-08-02:
https://github.com/scooterhacking/scooterhacking-issues-public/issues/93#issuecomment-2181459648
Explains why downgrading is not possible.

abdi2342, 2024-08-03:
why i cant tuning the xiaomi scooter 4 pro plus
i tried so much  but i cant get it tuning!!!!!

IAMFreehandZ, 2024-08-04:
> why i cant tuning the xiaomi scooter 4 pro plus
> i tried so much  but i cant get it tuning!!!!!

Not supported

LarsMde, 2024-08-04:
> [#93 (comment)](https://github.com/scooterhacking/scooterhacking-issues-public/issues/93#issuecomment-2181459648) Explains why downgrading is not possible....

Well, if it is not for security reasons, but only for the compatibility of the tool with the firmware, I would like to ask where I can find the source code of the SHFW firmware so that I can compare last year's with the current one - maybe I can find the error.
In the GIT I find a bunch of binaries of various original firmware, but nothing about SHFW.

BastelPichi, 2024-08-04:
> Well, if it is not for security reasons, but only for the compatibility of the tool with the firmware, I would like to ask where I can find the source code of the SHFW firmware so that I can compare [...]

SHFW is not Open Source/Source Open and very, very likely never will.

LarsMde, 2024-08-04:
Paulederbaus wrote on [comment93](https://github.com/scooterhacking/scooterhacking-issues-public/issues/93#issuecomment-2181459648) "The best way to get rid of them is to help us fix them."

I offer nothing else. Without looking at the source code or parts of it, I cannot help to fix the error.
So I can only hope that the operators of the hobby project will find the error themselves and that I will then donate a few thalers for the great work, because there is a lot of commitment involved.

Angelo1976ba, 2024-08-15:
Hello, i have the same problem ,battery external Is full charged
I have E22E change region in E45
[image]

lucienne3-web, 2024-09-02:
What are the chances that this problem will be fixed in the next release? Thanks

paulederbaus, 2024-09-19:
I contacted you via discord

closed 2024-09-20 by lucienne3-web

LarsMde, 2024-09-21:
> I contacted you via discord

Sorry, but there is no request on discord.

lekrsu, 2024-09-21:
> > I contacted you via discord
> 
> Sorry, but there is no request on discord.

It was sent to the creator of the issue report and it's already marked as closed.