scooterhacking/scooterhacking-issues-public #101: BUG: My NINEBOT E2 cant downgrade DRV and install SHFW

Alinus187, 2024-07-06:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Other

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

none

### SHFW Version

none 

### Scooter model

Ninebot Es2, Es3, Es4

### Description of the Bug

my scoter E2 cant downgrade DRV and install SHFW please help

### Steps to Reproduce

1 install ScooterHaking Utility 2 select my scooter 3 go to flash   3 press install/update SHFW (does not work) or attemp DRV downgrade(does not work as well

### Expected vs. Actual Behavior

I wish that this problem can be fixed

### Contact Information

totalin.

Nyami11, 2024-07-06:
It's not a bug. Ninebot E2 series isn't supported by SHFW, ans probably never will be.

paulederbaus, 2024-07-07:
Please send a screenshot of the info tab, but I guess Nyami is right

closed 2024-09-19 by paulederbaus