scooterhacking/scooterhacking-issues-public #102: Feature: Custom ways to enable Profiles

momochqn, 2024-07-14:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

UI / UX

### Priority

medium (upcoming releases)

### SHUT Version

2.3.0

### SHFW Version

0.3.7

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

There are already some ways to go into profiles that are known by almost everyone. It would be nice to be able to make own combinations. As example picking the buttons setting a time to press and more. So it‘s more hidden.

### Contact Information

@m9mo

Nyami11, 2024-07-14:
If you really want to make it hidden, just switch with utility. SHFW barely fits on the ESC, so I don't think Paul wants to waste space and his precious time on it.

nopbxlr, 2024-07-17:
Hi, I was actually working on a custom combination profile switch. I must get back to it. I had already made a UI prototype and started the SHFW code.

momochqn, 2024-07-17:
> Hi, I was actually working on a custom combination profile switch. I must get back to it. I had already made a UI prototype and started the SHFW code.

That sounds nice. Do you have it public anywhere for viewing? Is there any way for me to stay updated?

nopbxlr, 2024-07-22:
It is now going along again, nicely. It is close to ready for a future release :)

momochqn, 2024-08-05:
> It is now going along again, nicely. It is close to ready for a future release :)

Hey, i just wanted to ask how far the project is...
When do you think will it be released?

BastelPichi, 2024-08-05:
> > It is now going along again, nicely. It is close to ready for a future release :)
> 
> Hey, i just wanted to ask how far the project is... When do you think will it be released?

There is not going to be any ETAs, if everything is ready, it will be released with the next major SHFW version.

closed 2024-08-05 by momochqn

nopbxlr, 2024-08-07:
> > It is now going along again, nicely. It is close to ready for a future release :)
> 
> Hey, i just wanted to ask how far the project is... When do you think will it be released?

The implementation seems completed. Unless a bug arises, it should be in the next update