scooterhacking/scooterhacking-issues-public #103: Max ebrake voltage

Nyami11, 2024-07-15:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Performance

### Priority

low (take your time)

### SHUT Version

.

### SHFW Version

.

### Scooter model

Ninebot G30

### Description of the new feature

Implement a maximum voltage setting for regen braking to prevent tripping a battery's overcharge protection. Maybe add a second lower voltage setting, above which it would linearly reduce the brake amps until it goes over the other setting. Maybe you could also add a simular setting for overdischarge protections too. 

### Contact Information

Nyami11

Nyami11, 2024-07-15:
Also it would be good if it would throw some error, or turn on the overheat warning led, if the voltage is close, or above the set voltage

NormalHuman-Anything, 2024-10-07:
with a loud beep :>

PureComedi, 2024-10-07:
I think if it would be possible to have the brake section be just unified into 3 configurable values, brake strength, maximum amperage and maximum voltage, that would not only save a bit of space (i know of the shfw split but still) and would remove confusing stuff for new users