scooterhacking/scooterhacking-issues-public #104: BUG: My Ninebot D38e can't install SHFW

Susjed, 2024-07-15:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

SHUT 2.5

### SHFW Version

SHFW 3.7.2

### Scooter model

Ninebot G30

### Description of the Bug

won't let me install SHFW and my scooter isn't a G30 it's a D38e
[image]
[image]

### Steps to Reproduce

go to tools
click install SHFW

### Expected vs. Actual Behavior

SHFW installs and I can mod my scooter
SHFW didn't install and I can't mod my scooter

### Contact Information

susjed

KingBro213, 2024-07-15:
The message says that there is no public release, so a SHFW version for it exists, probably in Prerelease Testing. Ask other D-Series users (ideally d-series prerelease testers) if they have the same issue.

Nyami11, 2024-07-15:
We have no D and F series prerelease users. D series should be supported. I DM-ed Paul a few day ago with it, because someoe else asked with the same issue, and Paul said that BLE is corrupted. This is the third user I see ask about the same issue with the same BLE version and some 3.0.1.x DRV

closed 2024-09-19 by paulederbaus