scooterhacking/scooterhacking-issues-public #105: BUG: Reflasher's dialog box .NET core downloader still not working

PureComedi, 2024-07-24:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Other

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

.

### SHFW Version

.

### Scooter model

Ninebot G30

### Description of the Bug

The button does not download the dependency 

### Steps to Reproduce

Try to press the "Yes" button on 
[image]

### Expected vs. Actual Behavior

The software downloading, the software didn't download 

### Contact Information

purecomedi

BastelPichi, 2024-07-24:
Interestingly enough this works on some systems. On a fresh windows install in VirtualBox the Button works fine.

PureComedi, 2024-07-24:
There is 100x more cases of it not working than it working, it sorta became a norm when installing reflasher to suggest people to download it from the forum post