scooterhacking/scooterhacking-issues-public #108: BUG: Automate VLT LTGM not working with VLT LTGM+ DRV187 G30D & II

oopps24, 2024-09-04:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Other

### Severity of issue

low (asthetic, minor issue)

### SHUT Version

all

### SHFW Version

VLT DRV187 G30D & II

### Scooter model

Ninebot G30

### Description of the Bug

Hello
Automate VLT LTGM not load VLT LTGM+ for Ninebot G30D & II. On Xiaomi 1S and Pro2 it is working.
Is that a bug or not compatible? Isn't it compatible, give it maybe a chance that you update this feature?
Best regards

### Steps to Reproduce

1. Autoconnect Scooter (VLT LTGM Automate is activated) 
2. Automate is complete load and finished
3. VLT LTGM+ Button is not activated
4. I need to slide manuell the button to activate VLT LTGM+ on Ninebot G30D & Ii

### Expected vs. Actual Behavior

I thought it would activate automatically.
Is not activated after automate load.

closed 2025-02-10 by paulederbaus

rilup-1, 2025-02-11:
Why is that report closed? Bug is not fixed.

Z1mmb0, 2025-02-11:
you closed it 😄

paulederbaus, 2025-02-11:
Has anything been changed on the firmware side? New register or anything?

rilup-1, 2025-02-13:
is currently being clarified

nopbxlr, 2025-07-19:
Hello,

I'm writing to inform you about the upcoming removal of native support for VLT LGTM, along with other features such as automation and German Maneuver. These changes will take effect with the public beta release of App V3. App V2 will no longer receive updates but will remain usable for users who rely on these features.

This decision stems from a shift in how the app handles vehicle protocols and register maps. Rather than hardcoding these elements, App V3 will now fetch and cache real-time, model-specific configuration files. As a result, features not present in the stock firmware will fall outside the app's native support scope. However, this change enables better and faster support for each individual vehicle model.

That said, you will still be able to load your own offline packages, and we’ll be mirroring the current upstream packages on GitHub. With just a few simple modifications to a JSON file, you can re-enable support for VLT LGTM and even integrate broader custom firmware functionality.

More information will become available as the new version matures.

@VooDooShamane – let me know if you’d like details to help prepare for the switch to the new app version.

I hope the reasons behind this compromise are clear and that the benefits are evident. For any further technical questions, the ScooterHacking Assembly on Telegram remains a great place for both users and developers to get support.