scooterhacking/scooterhacking-issues-public #109: BUG: Can turn off scooter while locked thus unlocking it

arths31, 2024-09-05:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Other

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.6.1

### SHFW Version

3.9.0

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

You can turn off the scooter while it is in lock state thus unlocking it and making it driveable again.

### Steps to Reproduce

1. Connect to scooter using SHUT
2. Go to Tools tab
3. Enable lock
4. Turn off scooter by double pressing power button
5. Scooter is unlocked

### Expected vs. Actual Behavior

We should not be able to turn off scooter using power button when it is in lock state, just like with stock FW

### Contact Information

Telegram: Arths

Nyami11, 2024-09-05:
A hotfix was released.

arths31, 2024-09-06:
Problem fixed

closed 2024-09-06 by arths31