scooterhacking/scooterhacking-issues-public #111: BUG: Weak and jerky e-brake. Brake boost doesn't work at all. Pro 2 on 3.7.x and up.

fruitghost, 2024-09-06:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.6.1

### SHFW Version

3.9.1 and 3.7.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

The e-brake is really weak, it barely does anything at higher speeds and makes the wheel vibrate a lot on lower speeds as if the coils were turning on and off every few degrees. Brake boost doesn't work at all and changing brake current has very little effect. No amount of fiddling in settings fixed the issue.
It started in 3.7.1 (after acceleration and brake boost update) and is present in newest 3.9.1 shfw. It did not happen on 2023 firmware which i believe was <0.3.6. It seems that not all Pro 2 are affected but there are other reports showing the same behavior on both Xiaomi and Ninebots.
Attached video showing the vibrations. Keep in mind its louder and more jerky in person than it seems on video and I was able to easily move the scooter while ebrake was fully pressed.

### Steps to Reproduce

https://github.com/user-attachments/assets/6b750d96-a2cd-4645-b494-015f7bea3896

https://github.com/user-attachments/assets/9baa0a87-4284-4013-81b2-40cb5a90c7ff

### Expected vs. Actual Behavior

It should brake like stock or 2023 firmware.
It barely brakes. Vibrates a lot. Brake boost doesn't work.

Z1mmb0, 2024-09-06:
Confirming this behaviour. 
But i'd say the weak motor might be the key problem here. Or better said its higher kv-value. It looks like the motor is already pushed pretty hard to apply enough torque for braking without burning the phases long term.
Would be interesting to see how good the e-brake would be on a pro2 with g30 hardware.

noyina, 2024-09-07:
Can confirm the same issue and noise while breaking at low speed on a G2 with a UNAH motor.

paulederbaus, 2025-02-09:
Does the issue still exist in 3.10.1?

Z1mmb0, 2025-02-11:
i'd its fixed with the introduction of minimal brake speed. jerky when set lower than 3-5 kph. totally acceptable in my opinion

fruitghost, 2025-02-15:
@paulederbaus Unfortunately the issue is not resolved. Just updated to 3.10.1 and it behaves the same. Is there a way I can send you any diagnostic data? Or perform any tests?

@Z1mmb0 That just part of the problem and increasing minimal breaking speed only hides the jerky motion as it increases in frequency and is not noticeable. It was about weak ebrake and Brake Acceleration **completely** not working. Previous firmwares had it buttery smooth to the very 0km/h.

I think it's something deeper since even brake curve set to 60A doesn't slow down the scooter as much as older firmware but it sure kicks the wheel like a spring when it comes to a stop. So that high current **is** applied but imo it cuts out every few degrees. 'm365 Tools' charts report amps just as set in Utility.

oadq, 2025-06-12:
@fruitghost What settings are you using? Can you enable BMS emulation and show the current graph?
Also maybe try my settings (only on a Pro 2): 0% brake boost, 80A brake curve, 80A brake overshoot current

oadq, 2025-07-11:
@fruitghost What settings are you using? Can you enable BMS emulation and show the current graph?

Another thing is that "older firmware" (0.3.6 and older) increased your brake current by about 57% on Xiaomi.
On newer firmwares, your brake current is reduced at lower speeds.

Ngsfwatm, 2025-07-13:
Can you guys please please make the MI 4 lite 2nd gen work🙏

fruitghost, 2025-07-27:
@oadq 
[image]
[image]
[image]
Here are settings and two chart screenshots, I think the cropped one is with BMS Emulation set to 10S (?) although I don't know what does it change and other one with it off. Sped up to 25km/h and started braking with e-brake lever (this model has 2 levers).

> On newer firmwares, your brake current is reduced at lower speeds.
So was the old current shown incorrectly? Because I kept the same values in the app between versions. Only behaviour changed. And unfortnately as I mentioned earlier, higher current shows itself in a form of "springy" wheel, the force is there (even dangerously high) but cuts out in intervals shown in the first video (audio required).

Please inform me of more ways I can provide debugging data. The disc brake needs adjustment/drying every time it gets wet so lack of ebrake is quite a big loss for me :/
But also huge thanks to SH team for the features added in each version.