scooterhacking/scooterhacking-issues-public #114: BUG: Modes on G2(dont know if other models are affected)

Dervollzeittuerke, 2024-09-07:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

UI / UX fix

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.6.1

### SHFW Version

3.9.1

### Scooter model

Ninebot G2

### Description of the Bug

Scooter is Stuck in a Mode(ECO,DRIVE,SPORT)
Randomly while Driving

### Steps to Reproduce

Happens Randomly when driving or setting someting over SHUT
A fix i found out:
Go to Modes
Activate Sport mode only
Deactivate Sport mode only
Now it should work again

### Expected vs. Actual Behavior

Double press to Change mode but nothing happens

### Contact Information

Discord: e91_330d

paulederbaus, 2024-09-19:
@nopbxlr Please test when you have time.

Umba-SHU, 2024-10-03:
I would like to add to the description oif this bug as I have experienced it many times on numberouse scooters.

The bug is basically as described, BUT the workaround mentioned in this bug report is only a temporary worakound. 
Eventually the scooter would again get stuck in a mode again, even after toggling the "Deactivate Sport mode only" on and off. 

So there is NO actual workaround and I would like to ask to increase the serverity / priority of this bug as it affects every single G2 after SHFU is flashed onto it.

nopbxlr, 2024-10-04:
Will check it out when I have the time, putting it on my priority list

Umba-SHU, 2024-10-05:
Thank you very much, for looking into it.
I have a few fellow SHU users atm whom I switched to SHFU on G2 struggling with this issue now. Please post an update whenever possible. TBH I am not sure what was the purpose of this option anyway (to disable any other modes) especially while there is a very viable (and flawlessly working) option to boot on startup to the last mode used. 
Maybe this option might be even removed from the future releases of SHU (or at least temporarily disabled for the sake of quick fix and compatibility) ?
 
> Will check it out when I have the time, putting it on my priority list

Umba-SHU, 2024-10-08:
Adding to the description of the bug, I think I have found a workaround.  
If to choose in **"Set mode when entering profile"** anything but LAST (i.e. ECO, D, S), then the bug seems to not affect the usability.

closed 2025-06-23 by oadq