scooterhacking/scooterhacking-issues-public #115: BUG: Locking up the scooter makes the dash unresponsive

TastyTom79, 2024-09-07:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Other

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

SHUT 2.6.1

### SHFW Version

SHFW 3.9.1

### Scooter model

Ninebot G2

### Description of the Bug

Switching the scooter to lock an back to unlocked, makes the dashboard/main-button, throttle and e-brake unresponsive.
Scooter has to turn of by inactivity.

Side note: the lock functionality does not seem to do anything but disable the functionality of the above mentioned things.

### Steps to Reproduce

1. Start the scooter and connect it with SHUT
2. Use the "lock" funtionality
3. Use is again to unlock

### Expected vs. Actual Behavior

Expected:

-  Scooter shall work as before the lock procedure

Actual:

- Scooter becomes unresponsive (Throttle, E-Brake and Main-Button dont work anymore)

### Contact Information

tastytom - Discord

paulederbaus, 2024-09-19:
@nopbxlr Please test when you have time.

Umba-SHU, 2024-10-03:
I can confirm that the bug is still there and it's really annoying. It puts off unknowing people from using SHU and Ninebot scooters in fact. I had met a guy today who decided not to go for Ninebot G2 with SHU when this happened, we even had to open the scooter to disconnect the power line (we didn't knopw about the timeout). It just looks as a severe FREESE to the user.

nopbxlr, 2024-10-04:
The lock function is not usable as it was before, and thus will be removed from the app for G2 for now. Sorry! Should've paid more attention

SchattenWolf2008, 2024-11-26:
Hey I have also noticed this on the G30D II. However, I was able to just turn it off by the app. Its a bummer that you cannot use this to lock the scooter, or am I getting it wrong?

TastyTom79, 2024-11-26:
For G2 this actually doesnt work but one can still use the official NB App to use the CodeLock so its not to bad, dont know about G30 D II tho

paulederbaus, 2025-02-09:
Lock functionality hidden for g2

closed 2025-02-09 by paulederbaus