scooterhacking/scooterhacking-issues-public #117: BUG: 48V doesnt lead to expected speed increase on pro2

Z1mmb0, 2024-09-09:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

high (major functionality broken, no workaround)

### SHUT Version

2.6.1

### SHFW Version

3.9.1

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the Bug

When using CFW or old shfw versions (<0.3.5) you would expect the scooter going 31 kph on the 36V battery and approx 40kph on a 48V battery. 

at least since shfw 3.6.x you cant go faster than 28kph (without Fwk or the now introduced overmodulation) with 48V. 

with overmodulation you get like 33 kph. 
with Fwk around 40. 

Calculating from bare kv values the scoot should be way closer to 40kph. e.g. CFW reaches 38kph. 

### Steps to Reproduce

1. replace 36V battery with 48V
2. go for a ride
3. compare speeds

### Expected vs. Actual Behavior

Expected: 
48V should increase base scoot speed limit by 20-30% 
actual: 
36V system on cfw or older shfw releases without Fwk was just as fast as 48V on current shfw

### Contact Information

dc: zimmbo

paulederbaus, 2024-09-19:
Please show your settings. Prerelease didn't show this issue.

Z1mmb0, 2024-09-19:
Hello Paul, thanks for having a look. 

DPC, quadratic 25A, w/o overmodulation, w/o Fwk, PWM freq 16kHz. 

Quick recheck of the performance result: 
31kph peak speed at 50+V. 

[image]

I'm used to achieve these speeds with a 36V system on CFW. Shfw behaves different since the introduction of Fwk back in 0.3.5 or so.

lekrsu reacted to my ticket when i opened it. from his pov i expect to much from a 48V and suspects my battery dropping too much in voltage.

Z1mmb0, 2024-09-19:
On the other hand, this is what enabling overmodulation + fieldweakening (see parameters in screenshot) looks like. I'm happy with that and appreciate the amount of work behind the introduction of this stuff. 

[image]
[image]

paulederbaus, 2025-02-09:
Does this issue still exist with 3.10.1?

Z1mmb0, 2025-02-11:
technically - yes. turning off the features still results in a weirdly slow scooter. but practically who cares.

personally speaking, so far i arranged myself with the behaviour as several updates with good features were released in the meantime. e.g. overmodulation

after the update to 3.10.1 it looked like the the (torque) bottleneck increased. Definately the top speed went down from 40 to 37ish kph. So i raised the Iq limit to 65 A to regain previous speeds.
Note im aware this might also just be result from the fast changing conditions in winter. Its hard to always isolate the impacts. 

so far what i can see: with all features enabled, the scooter seems to pull 1000-1200 watts to approximate the 40 kph area. To me this looks reasonable if the readout actually includes Field weakening current as well.

Unfortunately the wattage is the metric i didnt consider when reporting differences between the old cfw and shfw.

closed 2025-02-11 by Z1mmb0