scooterhacking/scooterhacking-issues-public #119: Feature: CC with turning switch on G2

Figure-itout, 2024-09-16:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Other

### Priority

low (take your time)

### SHUT Version

n/a

### SHFW Version

n/a

### Scooter model

Ninebot G2

### Description of the new feature

G2 has turning lights build in. The stock firmware give you the option to enable cruise control by hold one of the buttons for a few seconds.
It would be nice if SHFW also support cruise control by holding the turning button for 3 seconds (or adjusting seconds with SHUT).

### Contact Information

Discord: ChaChaCha

CatInHat83, 2024-09-16:
Adding additional features for the aforementioned buttons would also be nice. 

For example: holding the turning button for x seconds could toggle Field Weakening on the go or switch between profiles/modes.

paulederbaus, 2024-09-19:
@nopbxlr Can you check how the input is detected in the esc?

Figure-itout, 2024-10-06:
@paulederbaus Is there a tutorial how to sniff?
I've got an Arduino sniffing the data from esc/controller to dashboard. Maybe I can help in some way.
But know the max g2 has 2 pcb's in the dashboard. I don't know if the turning switches past the dashboard to the esc.