scooterhacking/scooterhacking-issues-public #120: BUG: No throttle or headlight

ahaas25, 2024-09-19:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Crash

### Severity of issue

critical (app crash, data loss)

### SHUT Version

2.6.1

### SHFW Version

SHFW 3.9.1

### Scooter model

Ninebot E20, E25, E45

### Description of the Bug

Scooter has no response to throttle despite throttle curve being set. The headlamp does not work either (The LCD flickers and the headlamp flashes on and off, see video). Brake still works.

This was not an issue on previous versions of SHFW.

Is there a way to easily downgrade to the earlier version? Thanks!

https://cdn.discordapp.com/attachments/851989151922520065/1286119908052308008/PXL_20240919_000100556.mp4?ex=66ecc001&is=66eb6e81&hm=5ea83edc246d0919332905bd6b4048b0287b0ede314a60e24ae299021f73e9af&

### Steps to Reproduce

1. Install SHFW on Ninebot e22
2. Attempt to ride it -> Does not accelerate
3. Turn on headlight -> Flashes

### Expected vs. Actual Behavior

1. Install SHFW on Ninebot e22
2. Attempt to ride it -> Motor engages
4. Turn on headlight -> Headlight turns on

### Contact Information

poe on Discord (In the SHFW discord and also DM'd you)

paulederbaus, 2024-09-19:
Please check if "Regular-Base255-v3.10.0_v2_11" solves your issue.

lucienne3-web, 2024-09-19:
the same problem occurred to my E45

lucienne3-web, 2024-09-19:
where can i find Regular-Base255-v3.10.0_v2_11 ??

lekrsu, 2024-09-19:
> where can i find Regular-Base255-v3.10.0_v2_11 ??

**You** can't. It was only added for one scooter.

ahaas25, 2024-09-19:
Scooter is usable again. All functions work as intended. Field Weakening only seems to get me to 35 kph instead of the 43kph it used to on flat ground, but I'll open a new bug for that

closed 2024-09-19 by ahaas25