scooterhacking/scooterhacking-issues-public #121: BUG: Field Weakening not as strong

ahaas25, 2024-09-19:
### Scooterhacking Tool

Utility (SHUT)

### Type of issue

Crash

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

2.6.1

### SHFW Version

Regular-Base255-v3.10.0_v2_11

### Scooter model

Ninebot Es2, Es3, Es4

### Description of the Bug

Field weakening only accelerates me to 35 kph on flat ground. On SHFW v0.3.6 I was able to accelerate to 43 kph. Free spinning my scooter wheel also only gets 35 kph, whereas on SHFW v0.3.6 it could get up to 50 kph. I made sure there is no speed limit enabled in the SHFW.

### Steps to Reproduce

1. Enable field weakening
2. Configure it
3. Ride scooter at full throttle -> get to 43kph eventually

### Expected vs. Actual Behavior

1. Enable field weakening
2. Configure it
3. Ride scooter at full throttle -> get to 35kph and feel scooter cut power

### Contact Information

Poe

Z1mmb0, 2024-09-19:
reset your parameters, reinitialize and try again

ahaas25, 2024-09-20:
Tried this, no luck. Seems like field weakening is not as strong. Scooter barely gets to 35kph and any incline slows it down. Whereas before it could handle 35kph up moderate inclines

lekrsu, 2024-09-20:
Look over your settings or link them here.

ahaas25, 2024-09-24:
[image]
Here are my settings. Setting the max amperage to 15A didn't help either. I can feel field weakening kick in around ~32kph, but at ~33kph I feel the scooter cut power like a speed limit. I have the speed limit set to 100 KPH.

Nyami11, 2024-09-24:
Turn on expert view at the top, set initial to 0 and variable current to 1000. Also if you just set speed limit to 0, it turns it off...

ahaas25, 2024-09-24:
That's my bad, that fixed it!

closed 2024-09-24 by ahaas25