scooterhacking/scooterhacking-issues-public #123: Feature: Auto braking speed control

DienoX, 2024-09-23:
### Scooterhacking Tool

Utility (SHUT)

### Type of feature

Other

### Priority

medium (upcoming releases)

### SHUT Version

Shut

### SHFW Version

SHFW

### Scooter model

Xiaomi Pro, Pro2, 1s, essential, 3

### Description of the new feature

Please add options to adjust auto braking speed.

The ability to set auto braking speed will enable more smooth and efficient management of battery consumption. Additionally, it will allow you to easily push the scooter, e.g. through a pedestrian crossing.

### Contact Information

Discord: Lutris.

lekrsu, 2024-10-09:
Brake min speed should affect autobraking as well. Isn't it?

DienoX, 2024-10-09:
I would prefer two separate settings. One is the speed at which auto braking stops working. The second is the setting in which the scooter stops engine braking regardless of auto braking.