scooterhacking/scooterhacking-issues-public #124: Feature: Utilizing Unused Connectors on Scooters for LED & Star-Delta Mod & more

Ben-Star-Delta, 2024-09-30:
### Scooterhacking Tool

Firmware (SHFW)

### Type of feature

Performance

### Priority

medium (upcoming releases)

### SHUT Version

/

### SHFW Version

/

### Scooter model

Ninebot G30

### Description of the new feature

_Dear **ScooterHacking** Team,_

I have noticed that there is an unused 3-pin connector on the G30D ESC. It would be fantastic if this connector could be utilized for additional functions in future firmware updates, such as:

> LED Strip Connection: Enabling the direct connection of LED strips to the 3-pin connector would allow users to add lighting effects to their scooters. This would make the installation and control of LED strips much easier without the need for additional wiring or modifications.

> Star-Delta Mod: For the Star-Delta mod, the 3-pin connector could be used to control an Arduino relay, which switches on or off at a selected speed. This would provide users with a simple and efficient way to implement the Star-Delta configuration.

Furthermore, I’ve noticed that **other models**, such as the G2, also have **unused pin**s. It would be great if these possibilities could be explored and made available through firmware updates across different models as well.

I believe these enhancements would greatly improve the user experience and make the scooters even more versatile. I would greatly appreciate it if you could consider adding this feature.

Thank you for your excellent work and continuous support of the scooter hacking community!

_Best regards,
[Ben Star-Delta]_

### Contact Information

0.moon_light.

PureComedi, 2024-09-30:
A way i could see this implemented in a developer and user/functionality friendly way is introducing an SHFW patcher/plugin/special features thingy that would be able to remap certain ports to certain (basic) features. The user would be able to see the maximum amount of flash memory they can use and the current size of the firmware that they have. All this imo would allow for less crucial features that the average user may not ever need but some certain niche-er group would find useful.

If this is going to be implemented I'd also recommend moving a few current features into the extras section, just even considering how many of shfw's features are unused by most average users.

nopbxlr, 2024-10-02:
A reply to this latest comment - this could be done in theory way simpler with the division between SHFW core/SHFW lib as discussed on Discord or the ScooterHacking Assembly. However, with the gain of space this very division would give, the feature selection wouldn't be necessary

About the feature discussed in this thread - I do not have any specific opinion to give as of now. It could be considered when the time is proper. But the priority is definitely given to most used features, quality of life improvements, and right now, most importantly, the SHFW split in two parts

Ben-Star-Delta, 2024-10-04:
I fully support prioritizing the split of SHFW, as it will bring many benefits. However, I don't want the use of the unused connectors on the ESC to be forgotten. These offer great potential for future extensions and shouldn't be overlooked in the long run.

nopbxlr, 2024-11-14:
Currently working on SHFW V4 (fully compiled from scratch, no ninebot code), and today, had to deal with said GPIOs. It is way easier to implement it with the new code. This feature will then make it, unfortunately not in the current V3 but when V4 will hit the release state. I do not have any ETA to share, unfortunately. This takes time