scooterhacking/scooterhacking-issues-public #126: BUG: Speed limit setting is not working with high performance settings on G2

Umba-SHU, 2024-10-08:
### Scooterhacking Tool

Firmware (SHFW)

### Type of issue

Performance issue

### Severity of issue

medium (affects usability but has a workaround)

### SHUT Version

SHUT 2.6.1

### SHFW Version

Regular-Base170-v3.9.1

### Scooter model

Ninebot G2

### Description of the Bug

BUG: Speed limit setting is not working with high performance settings on G2

### Steps to Reproduce

1. Set speed limit, e.g. 45 km/h
2. Enable Field Weakening with settings like in the screenshots (or any higher performance settings than these) 
3. Enable PMI Overmodulation 
4. Enable Zero Start
5. Put rear wheel in the air and press accelerator 

The wheel spins up so fast that it goes far beyond speed limiter setting and keeps spinning at that speed while accelerator remains pressed,  

PS Have tested this only with "wheel in the air" can't confirm it still happens with load. But assuming the limit should be applied in both cases. 
[image]
[image]

### Expected vs. Actual Behavior

Expected: Set speed limit setting should override the acceleration speed at all times.
Actual behavior: Speed limiter not applied.

Workardound: at more mild settings of FW the speed limiter is working. It seems that the motor surpasses the speed barrier (with high performance settings) so fast that the speed limiter fails to register the barrier it is set to, and is not applied at all then. 

### Contact Information

@ilyasamui

lekrsu, 2024-10-09:
You don't use field weakening in-air on any system.

closed 2024-10-09 by lekrsu

Umba-SHU, 2024-10-09:
> You don't use field weakening in-air on any system.

Thanks and good to know. I am not sure I have seen the warning not to try it in-air in the app

Umba-SHU, 2024-10-09:
Thanks maybe worth adding this warning to the SHU in the  (i) section for FW option